umrover/mrover-ros #53: Robot doesn't move in Gazebo sometimes

rbridges12, 2022-06-19:
Sometimes when you start one of our launch files, everything will appear to start correctly except the robot will not move in response to velocity commands. This seems to happen with no pattern, but usually around 1/4 of the time, although that's just on my main system. 

I've seen it happen on several different systems, and other members have seen it as well. When using a Docker container, it seems to happen much more frequently (about 90% of the time) but this may have been a fluke.

Initially it seemed like an issue with the Gazebo 6w drive plugin, but after spending a little time debugging, it seemed like it was a deeper issue.

niwhsa9, 2022-06-20:
removing sim-gap atm because its currently unclear if this is a simulator issue or a bug in nav

niwhsa9, 2022-08-16:
definitely a sim-gap, readding the label see #77

closed 2022-08-21 by niwhsa9