umrover/mrover-ros #59: Cache on perception

qhdwight, 2022-06-30:


niwhsa9, 2022-08-17:
this needs to happen within navigation node right?

qhdwight, 2022-08-20:
@ankithu what do you think, I figured this would be a perception thing

ankithu, 2022-08-20:
Trying out the response by email thing again. Um I mean technically it can
happen in either and work exactly the same way obviously but like it’s more
of a question of what we think the role of the perception node actually is
in our system. Is perception meant to provide the best possible estimate of
what’s going on around us (in which case probably all the filtering and
stuff should happen on perceps end) or is it to literally report exactly
what it sees at any given moment (in which case it should be nav). Honestly
I don’t think I have a super strong opinion either way on this one, like at
the end of the day both are going to work so it seems like more of a code
philosophy question to me. Unless I’m mistaken and there is a
technical/controls advantage to doing it on one end or the other?

niwhsa9, 2022-08-20:
I've talked about this issue a lot with Carmen and Zack and we agreed it should happen in nav unless perception wants to start assigning probabilities to it's estimates (e.g. Kalman Filtering)

- Perception shouldn't be hallucinating or lying about an AR tag where it doesn't exist for the benefit of nav
- Nav system should be robust to false negatives because this is a reality of computer vision with or without filtering

Cacheing is not filtering, it's a mechanism for dealing with false negatives or no immediate visibility because nav can still use out-dated information to make logical decisions about how to progress. It's not perceptions responsibility to continuously provide that, it's nav's responsibility to hold onto it if they need it. 

In other words, even if we Kalman filtered detections so that we can give a tag location without an immediate sensor measurement, it would still make sense for nav to cache if they feel the likelihood of the current detection is too low to be useful. 

The perception microservice is a data provider, not part of core navigation logic (which is what cacheing is).

qhdwight, 2022-08-20:
Ok I am fine with having this in navigation then.

My only reservation is related to what Ankith said with this sentence: "Is perception meant to provide the best possible estimate of what’s going on around us (in which case probably all the filtering and stuff should happen on perceps end)"

Ideally perception would not publish tags that are just straight up wrong. An example would be a smaller service that relies on perception for correct tag info, we wouldn't want the caching logic twice (ig it could be in a separate library).

niwhsa9, 2022-08-21:
You're referring to false positives? Yes ideally perception doesn't do that.

niwhsa9, 2022-08-22:
i think the only situation where percep would be publishing tags that are 'wrong' is false positive detection, which I agree we should absolutely prioritize avoiding that. If the cache needs to be its own service thats fine, but no point introducing that overhead until we deem that necessary.

Aditya-Singhvi, 2022-09-15:
Updating issue with in-person discussions so far:

- AR tags are stored as immediate AR tags as they are detected in each frame; if an AR tag is detected in [n] consecutive frames, it becomes a persistent AR tag, and every subsequent reading is stored in its associated circular filter. 
- However, this system essentially drops the first [n] readings of an AR tag, effectively ignoring good data. To rectify this, we were considering implementing an "intermediate" AR tag stage: immediate AR tags are the ones seen in the current frame, intermediate AR tags are all tags seen in previous frame(s) and store all previous location data with them in a filter (but have been seen less than n times), and persistent AR tags are seen >= n times.

mberkowitz03, 2022-10-02:
Updated percep wiki (under update loop) for cache steps

PR #189 (2022-10-30): False positive detection on ArUco tags + PCL
## Summary
Closes #59 and #227 (merged in)

Adds a false positive checker over the tags we get from OpenCV using a hit counter. IDs have to be hit a configurable amount of time before they are published to the TF tree.

This PR also changes perception no longer keep publishing persistent tags. It is now navigation's job to cache and determine when a tag has expired.

We also use PCL to easily access the XYZ position with a better API. The old method had to memcpy and was ugly.

### Did you add documentation to the wiki?
Yes - under percep

## How wa [... cut]
closed 2022-10-30 by qhdwight