umrover/mrover-ros #70: Python format check to black is breaking stuff/not working properly

tabiosg, 2022-07-17:
I'm currently developing in tabiosg/cameras_ros and umrover/cameras, and I noticed two things:
1. Running ```black --check --line-length=120 src scripts``` or ```black --line-length=120 .``` passes on my computer, but it doesn't pass GitHub actions workflow style checker (not sure why?).
2. isort and black conflict with each other which is annoying (I'd assume this would prevent my code from ever passing the style checker, but I never made it that for so IDK).

Extra info:
black wants
```
from mrover.srv import (
    ChangeCameraMission,
    ChangeCameraMissionRequest,
    ChangeCameraMissionResponse,
    ChangeCameras,
    ChangeCamerasRequest,
    ChangeCamerasResponse,
)
```

while isort wants

```
from mrover.srv import (ChangeCameraMission, ChangeCameraMissionRequest,
                        ChangeCameraMissionResponse, ChangeCameras,
                        ChangeCamerasRequest, ChangeCamerasResponse)
```

IDK what labels this falls under but I'll just mention @niwhsa9 since he worked on #67

niwhsa9, 2022-07-17:
did you merge in origin/master into your dev branch? The CI job uses the .GitHub folder in your current branch, which I'm thinking is out of date.

Also, yeah we dropped isort for now because it can conflict with black. This can be reconfigured though so I'll deal with that later

niwhsa9, 2022-07-17:
just pushed a commit to your branch that passes CI. When I ran black on src/ it complained about the same thing as CI and fixed it. I wonder if theres a version incompatibility? What do you have for `black --version`? What command did you run to actually execute black locally?

tabiosg, 2022-07-17:
So I ran both ```black --check --line-length=120 src scripts``` and ```black --line-length=120 .``` locally. My ```black --version``` is version 19.3b0

niwhsa9, 2022-07-17:
The `black .` in the root dir of the package should've worked. I think this might be due to you having an older beta version since I'm on 22.6.1. I would give that an update with pip, checkout HEAD~1 on your branch, and re-run black to make sure it does what we expect.

tabiosg, 2022-07-17:
Kinda silly question but how do you do change your version of black?  
[image]

niwhsa9, 2022-07-17:
Looks like you need to run ~/.local/bin/black instead of just black since there's an old version of black on your path.

tabiosg, 2022-07-17:
Alright yea using the right version fixed it! I'll keep in mind to change my path and not ignore the fat yellow warning that shows up next time lol

closed 2022-07-17 by tabiosg