umrover/mrover-ros #84: Investigate IMU filtering offboard

niwhsa9, 2022-08-23:


niwhsa9, 2022-08-23:
https://pypi.org/project/inekf/

rbridges12, 2022-10-10:
We need to fuse/filter our raw IMU data offboard (on the Jetson rather than on the IMU's builtin processor) so we can tune it to give us a more globally accurate orientation.

rbridges12, 2022-10-10:
Here's a [library](https://ahrs.readthedocs.io/en/latest/) to look at

rbridges12, 2022-10-26:
The most commonly used filters in industry are madgewick, mahoney, and extented kalman filters (as far as I know)

rbridges12, 2023-06-08:
closing this because we didn't really make much progress on it this year, and the onboard filtering solution on the IMU doesn't seem to be much of a problem anymore, and in the future we're going to look into more correction based solutions rather than our own custom filtering.

closed 2023-06-08 by rbridges12