umrover/mrover-ros #86: simulator regression test

niwhsa9, 2022-08-23:
We need a sort of 'capture all' regression test that simulates the auton mission in gazebo (maybe a regular waypoint and a post) and this should get run as part of CI.

qhdwight, 2022-08-24:
Agreed

ankithu, 2022-08-24:
Yea I agree as well. Ideally there should be a way to run it without
needing to see what’s happening visually (headlessly)

niwhsa9, 2022-08-24:
Running headlessly is no problem. Just launch a gzserver without a gzclient. You can set the real time update rate to 0 to get it to run as fast as the CPU allows.

This should be a ROS test file. I'm envisioning something like:
- launch gazebo server and spawn the robot in a predefined world
- launch the auton nodes
- launch another node that has the test set. This one subscribes to nav to track the state transitions in addition to accessing the rover's ground truth position in the world. It then tests for various completion events (e.g. if state is X and pos is approx P). 

One important thing is that it has to have a configurable time out if it runs as part of CI. If it exceeds the timeout it should report test failure.

PR #141 (2022-09-24): rostest Integration and Regression Testing
## Summary
Closes #86 

I added a rostest that runs a very simple auton mission. It will act as a regression test and is required before merging into master. It can be invoked via `catkin test` (to run all) or just `rostest mrover integration.test`

I also refactored the way that courses are managed by making a custom `CourseService` class. It handles maintaining the dual state of having a topic for the course and also TF entries.

Some other differences:
- There is now a `mrover` user in the docker container. GitHub actions have to be run as root tho [... cut]
closed 2022-09-24 by qhdwight