umrover/mrover-ros #87: Surface normal cost maps

niwhsa9, 2022-08-23:
Lets try to build some cost maps using the surface normals. Here is some context on that which may or may not be helpful:
[image]

Lets see some visualizations to see what heuristics work well. We might want to do histogram bins as opposed to having the cost maps be the resolution of our point cloud, since this will also smooth noise out.

qhdwight, 2022-08-24:
ZED gives the normals very easily so this will be good.

@rbridges12 and I also talked about this, but in his lab they detected obstacles by looking at the normals as well - they will stick out horizontally. Still mark the ground based on what you described $normal(x,y) \cdot <0,0,1>$. Would have to be careful about the case where there is an obstacle with a flat top. Could also just give points that we think are obstacles a weight of infinity (so they are not even considering by A* or whatever)

I like the idea of binning the cost map (some form of kernel) since there will probably be some noise.

JasonLiu2002, 2022-09-20:
Update:

The ZED does give the normals easily, but the ZED ROS wrapper node does not publish the normal map. We've forked the repository here: https://github.com/umrover/zed-ros-wrapper and this must be cloned in the `src` directory of the Catkin workspace alongside the main `mrover-ros` repository.

To publish the normal map, we've added 3 more channels to the point cloud which is published to the `/zed2i/zed_node/point_cloud/cloud_registered` topic. The 7 channels are now, in order:
- x (float32)
- y (float32)
- z (float32)
- rgb (32 bits, each rgba channel taking up 8 bits/1 byte)
- normal_x (float32)
- normal_y (float32)
- normal_z (float32)

We've also changed the coordinate system the ZED uses to the OpenCV coordinate system, but will likely change this to use the ROS coordinate system to eliminate the need to translate between different coordinate systems.  Subscribing to the new 7-channel point cloud has been tested, and the normals appear to be correct. (Tested by printing out the normals, visualization has not been done yet)

Next steps:
- Edit `zed-ros-wrapper` to use the ROS coordinate system
- Design and implement the actual cost map calculations

JasonLiu2002, 2022-09-20:
@niwhsa9 what do you think about using the same ROS coordinate system for the ZED?

qhdwight, 2022-10-31:
Notes from 10/31/22
- Intermediate cost map node between zed ros wrapper and navigation that keeps a local data structure of points
- Each frame we get the transform between the last pose and our current pose
- Use this transform to update the existing cost map
- For new point cloud data that falls in the bins, evict the older ones and only use the newest one

niwhsa9, 2023-03-12:
not pursuing this for the time being

closed 2023-03-12 by niwhsa9