umrover/mrover-ros #98: TF exception handling in nav code

rbridges12, 2022-08-25:
Whenever we get an SE3 from the TF tree, there are several possible exceptions that can be raised and need to be caught. This is done in a couple places already in the nav code, but there are several places where TF tree lookups are happening with no exception handling (when either `rover.get_pose()` or `context.waypoint_pose()` are called). We need to add exception handling to these places, thinking carefully about what these errors mean and what behavior we want in those situations.

ankithu, 2022-08-25:
What’s the nature of these exceptions? Like rover.get_pose() will throw
because there is no rover pose in the tf tree? In that situation we’d
probably want to block until we get something right probably push some
error message too.

rbridges12, 2022-08-25:
there are 3 exceptions that I put :raises: tags for in the docstring of SE3.from_tf_tree(), so once that gets merged in you can check on them there. basically there's one for a frame not being in the TF tree, one for there not being a connection between the two frames in the TF tree, and one for when you have to look too far in the past or future to connect them.

Aditya-Singhvi, 2022-11-10:
**Closing issue after discussion:**

Exceptions are raised in the _from_tf_tree_ method. _from_tf_tree_ is called in 3 places:

1. 	rover.get_pose				
2. 	Course.waypoint_pose			
3. 	Environment.get_fid_pos  		This one already has try-catch for all 3 exceptions inside the method

In the first two cases, not being able to find a pose would be fatal, and we would want the code to crash — so no exception handling should be added / needed.

closed 2022-11-10 by Aditya-Singhvi