umrover/mrover-ros #102: Black style check working differently on Github vs local machine

jnnanni, 2022-08-28:
Code from PR #100 is not passing black style check on github but cloning the branch and running the same "black --check --line-length=120 mrover/src mrover/scripts" claims that no files would be changed.

niwhsa9, 2022-08-29:
This is with black 22.6.1 on my system. I'm getting the same thing as CI. 
```
ashwin:~/Documents/mrover-catkin/src$ black --check --line-length=120 mrover/scripts/ mrover/src
would reformat mrover/src/teleop/jetson/jetson_teleop.py

Oh no! 💥 💔 💥
1 file would be reformatted, 16 files would be left unchanged.
```

jnnanni, 2022-08-30:
I just formatted my code with 22.6.0 and pushed another commit that is also failing if you'd like to try that one. Could it be that you and CI are both running 22.6.1? My pip and the black github show that 22.6.0 is the latest release.

rbridges12, 2022-09-01:
I also have 22.6.0 and it passes fine for me

niwhsa9, 2022-09-01:
looks like version skew is the issue, lets all standardize on what the CI job uses: https://github.com/ricardochaves/python-lint-image/blob/master/CHANGELOG.md which is black 20.8b1

closed 2022-09-01 by niwhsa9