umrover/mrover-ros #126: Add requirements.txt

rbridges12, 2022-09-03:
Instead of having to install all of our necessary pip packages by hand, especially since some need specific versions, we should just make a requirements.txt containing all of those pip packages so we can install with pip -r requirements.txt

qhdwight, 2022-09-03:
Agreed this is needed. Can this be solved with `package.xml` out of curiosity?

rbridges12, 2022-09-03:
Sounds like a no based on this [post](https://answers.ros.org/question/376259/rosdep-install-specific-version-of-dependencies/). rosdep doesn't let you specify specific versions.

As mentioned in that post, maybe [this package](https://github.com/locusrobotics/catkin_virtualenv) could be a solution, but I'm not sure what the pros/cons of introducing a python venv into our system are.

qhdwight, 2022-09-06:
Solved by #123

closed 2022-09-06 by qhdwight