umrover/mrover-ros #142: get RTK working

rbridges12, 2022-09-08:
Our GPS units have support for RTK (check localization wiki for more info) but we haven't been able to get it working. This is mainly a matter of figuring out how to configure the firmware on each of the 2 GPS units so that RTK works.

niwhsa9, 2023-03-12:
RTK system with dual GPS configuration on the rover in addition to the base station can provide centimeter accurate relative positioning and accurate heading as well. This would suggest a localization strategy of replacing the current VO/IMU odometry frame with an RTK odometry frame. The primary benefit of this is that unlike the current Odometry frame, RTK drift is bounded over time. Namely, the transformation from the RTK frame to the map frame is subject to the noise level of the GPS (which is a couple of meters). This would allow us to do long distance planning and driving in the Odometry frame which currently is not possible. 

With a working RTK or DGNSS system, at least for waypoints with perception (gate and post) we should be able to operate entirely in the odometery frame which will eliminate the need for GPS filtering/smoothing and heading correction.

rahdal, 2023-04-02:
Doc with RTK resources/documentation:
[https://docs.google.com/document/d/1WzWrfvzHNGp2uje7ifqANK_043xySyMr2kfynFbP7ac/edit](url)

nehakankanala, 2023-04-03:
> Doc with RTK resources/documentation: [https://docs.google.com/document/d/1WzWrfvzHNGp2uje7ifqANK_043xySyMr2kfynFbP7ac/edit](url)

The link isn't opening for me

rbridges12, 2023-04-03:
yeah same

rahdal, 2023-04-03:
fixed

rbridges12, 2023-06-25:
# RTK TODO

## GPS Configuration
The GPS receiver has a ton of configuration options, so we have to find the best set of options for both the rover and base station. I think Rahul has mostly finished this, but I recently got a lot of experience with all these configuration options so I'll probably look it over and make sure there's nothing we need to change.

## GPS Driver
Once the GPS is configured, it'll be sending a bunch of messages over serial that contain all the information we want, so we need a program to read and parse those messages and then publish the data to ros. 

There is an existing ublox driver ros node from Kumar Robotics (linked in the resources document) that looks pretty extensive and seems to have all the features we need from a glance. I believe Rahul has been using it so far without any problems, but we'll need to test it thoroughly especially with sending and receiving RTCMs to make sure it works properly.

If that node doesn't work out for some reason, we can write our own, probably in python using the pyubx2 library (also linked in the resources document) which works really well in my experience.

## RTCM Transmission
The key thing we need in order to get RTK is the ability to send RTCM messages over the radio from the base station to the rover. These messages come from the base station receiver over serial in the form of binary packets, and they need to be transmitted to the rover receiver in the same form.

There are 2 main ways I can see us doing this:

1. simply read the packet from serial and put it in a ros message as an array of bytes or something, then send that ros message to the rover receiver node and have it sent over serial to the receiver. it looks like this approach is already mostly implemented in the ublox ros node, but we'll have to test it because I'm not sure if it works.

2. instead of sending it directly over the ros network, we can use the NTRIP protocol to send it separately over the network. This basically requires having an NTRIP caster running on the base station and an NTRIP client on the rover. This approach is much more common so there is a lot more documentation about using it, but in our case it adds a lot of complexity and overhead, so this is just a backup option.

## Manufacturing Ground Planes
so it turns out our receivers don't work very well unless they have some kind of flat metal surface directly beneath them, because of some complicated RF physics that I probably don't understand. we need to talk to mechanical about manufacturing a few of these for testing, and getting them mounted to the rover next year. we should probably do this sooner rather than later, because mechanical design cycle is starting soon, and RTK may not even work without the improved the improved reception quality from the ground planes.

## Dual RTK Heading
once we get basic RTK working well, we'll want to try adding another receiver to the rover with antennas mounted a different positions. this will give us two high precision position points on the rover, which will allow us to calculate heading.

we currently have a total of 3 GPS receivers and 2 antennas (I think?) so we could theoretically test dual RTK after only buying another antenna which is like 50 bucks. that probably isn't a great idea especially before CIRC but even after, and if this works we definitely want to buy at least one more receiver (~$200) so we should probably talk to the money people about when we can buy the new receiver and if we should buy it before testing it.

software wise there probably isn't much work to do here, because adding the other receiver is probably as simple as just running an additional driver node on the rover. I've read that the receivers can handle the differential heading computation by themselves, but I haven't verified that or seen it in the config. Even if we have to do it ourselves, it shouldn't be that hard to compute manually.

rbridges12, 2024-02-01:
we should close this in favor of the newer more specific issues, but maybe we need to transfer all this documentation somewhere else first