umrover/mrover-ros #148: Test networking setup with basestation and jetson

niwhsa9, 2022-09-13:
need to make sure we have a realistic setup

rbridges12, 2022-09-13:
I added configuration info to the wiki under ROS fundamentals

tabiosg, 2022-09-17:
@qhdwight do u want to set the ip of the jetson u have to 10.1.0.2 tomorrow for us? Check the image below to see the ips that we have set for all our devices.
[image]

Also a general comment for anybody reading this:
i kinda don't understand why the base station has a different network ID than everything else though (10.0.0 instead of 10.1.0) so it would be nice if someone knew why so we don't just have legacy code for no reason

qhdwight, 2022-09-17:
Yes I can do that

qhdwight, 2022-09-18:
Isn't 10.1.0.1 usually reserved for gateway? e.g. the radio?

qhdwight, 2022-09-18:
Also I think leaving stuff at 10.0.0.X is better

qhdwight, 2022-09-18:
Here is how you should set up the network with the jetson (client) + a computer/basestation (host).

Plug in an ethernet cable between the two. On the host run `sudo sysctl -w net.ipv4.ip_forward=1` and run `ip a` to figure out what your WiFi interface is called. It should be `wlo*` or something similar. Enable postrouting with `sudo iptables -t nat -A POSTROUTING -o <interface here> -j MASQUERADE`.

I configured the jetson to be `10.0.0.2` and have the mDNS hostname `mrover-jetson-2`. So you should be able to `ssh mrover@mrover-jetson-2` (or use IP if that fails). Once you are in SSH the Jetson should also have shared internet through your computer.

qhdwight, 2022-09-18:
The network configuration on the Jetson is handled via netplan. You can see the file at `/etc/netplan/01-netcfg.yaml`. That is where the static IP and such is set.

tabiosg, 2022-09-18:
Ok u need 10.1.0 because you need the first 3 numbers to be the same as the radios so they are all on the same network i think. We can test it but this is from my experience

tabiosg, 2022-09-18:
Also as a follow up, is there a way to make "sudo ip addr add 10.0.0.1/24 dev <ethernet interface here>" permanent?

We have been using that command to pull up air OS but it is a very temporary solution (the IP is forgotten as soon as the ethernet is disconnected). Also, the exact command we've been doing is "sudo ip addr add 10.1.0.7/8 dev <ethernet interface here>" so idk if that 8 vs 24 matters.

qhdwight, 2022-09-18:
Yeah you can do the same thing with netplan on the base station to make it permanent. See https://netplan.io/examples for more info. I would not recommend it for your laptop as it may screw up your ethernet for other things.

niwhsa9, 2022-09-18:
What no use nmcli con mod

qhdwight, 2022-09-20:
Close after I add documentation to WiKi?

tabiosg, 2022-09-20:
Yes sir

tabiosg, 2022-09-24:
Closed by #89

closed 2022-09-24 by tabiosg