umrover/mrover-ros #156: Refactor perception to use pointcloud for RGB data

qhdwight, 2022-09-18:
Currently we subscribe to two topics for processing - image data and also point cloud. They are separate topics & data structures and it is annoying to synchronize the two.

Instead just use the point cloud and try to read RGB data from there. Then pipe that into OpenCV.

yonMaor, 2022-09-18:
Hey, I'd like to help, but I'm completely new to the project.
Do you have more details about this issue?

Aditya-Singhvi, 2022-10-04:
1) Write method to convert PointCloud2 data to OpenCV image — make sure it conserves all data conserved by toCVCopy function 
2) Refactor cache code into more modular functions
3) Put everything in point cloud callback and don't subscribe to image anymore

qhdwight, 2022-10-18:
Handled by #189 now

qhdwight, 2022-10-24:
Unfortunately not possible without modifying the ZED ROS wrapper. It does not publish RGB data for points that dont have XYZ. It appears they publish image and point cloud on a separate thread too. Maybe in the future we can fuse both them with a custom wrapper.

closed 2022-10-24 by qhdwight