umrover/mrover-ros #163: Add ESW Debug menu to base station GUI

jnnanni, 2022-09-20:
As quoted from Guthrie:
"Create a debugging gui page that has a bunch of teleop and esw other data for arm and drive. Kind of like an esw teleop interface gui
E.g. let's say esw is debugging their arm code. They pull up the gui and they can see 1. Data being published to joystick and xbox 2. Data being published by drive callback and ra callback and 3. Data being published to individual joints for esw to read.
While you are at, also maybe include stuff that esw publishes to like drive vel data and drive state data and soon will be arm data too"

This page will be added as a new route and will have logging for any ROS topics that are relevant to ESW. Ask myself as well as @tabiosg and @sashreek1 for what data in particular.

jnnanni, 2022-09-25:
To do this, I would reccomend adding a json file to the gui/src/static folder that has a list of the ros topics that you want to subscribe to, then, in the file for the page, read in the json and have a v-for loop that creates the output and subscription on the GUI for each of them.

tabiosg, 2022-09-25:
Here I'll give a bit more details:

Wed like to see the joystick and Xbox controller data (currently exists as a msg/topic published by the gui)

Wed like to see the drive/arm control functions that are created from the joystick and xbox (i think this is published by jetson/teleop... either way its being listened to by esw)

Then wed like to see the data that esw publishes (esw currently posts odrive state and wheel current and velocity data)

closed 2022-11-05 by jnnanni