umrover/mrover-ros #187: Investigate LFS storage solutions

niwhsa9, 2022-09-30:
We currently have large files in the repository, namely the mesh files for the rover model used in our simulator. It's less than 1 GB of stuff but Github's default LFS (large file storage) has free usage limits and we are exceeding them. 

We want to see if there is another service that we can leverage so we can avoid paying for Github LFS. 

One potential path to investigate is if we can use Google Drive API. The basic workflow would be:
- Users clone the repository 
- They run a script e.g `python3 get_large_files.py` 
- Large files are downloaded from GDrive 

We would need to figure out how to handle permissions and the likes. Ideally, read-only permissions for downloading the files requires no credentials/keys/passwords. If thats not possible, is there a way to get per-user credentials (perhaps associated with UMich google drive login?). One thing that probably isn't acceptable is shared keys/credentials for the entire team, since that is likely disallowed by policy (not sure though). 

If not Google Drive, perhaps there are other services out there that we need to look into.

niwhsa9, 2022-09-30:
https://developers.google.com/drive/api

niwhsa9, 2022-09-30:
btw, if the API doesn't work out you might even be able to `curl` or `wget` the files just through Google Drive itself.

niwhsa9, 2022-09-30:
you may also be able to use githooks to automatically get the large files on checkout: https://githooks.com/ https://git-scm.com/docs/githooks

niwhsa9, 2022-09-30:
The folder to try this on is rover_descritpion/meshes. That folder contains the large files

niwhsa9, 2023-03-12:
we have a githook that cna do this now see #209 however this does not address how to version control files and additionally requires manual installation of the hook which will prevent this from working in CI. Further work needed.

rbridges12, 2023-06-08:
do we still want to keep looking into this? it seems like git LFS is working fine for us and doesn't cost that much, and we're losing our Google Drive space so we would need to find a different cloud storage integration

niwhsa9, 2023-06-08:
nah its fine, its $5 a month we can afford that.

closed 2023-06-08 by niwhsa9