umrover/mrover-ros #188: Fix ROS-Gazebo-Depth Camera Point Cloud Disparity

arschallwig, 2022-10-02:
Currently the Openni Kinect Depth camera we use to spoof point clouds in the perception stack uses a different frame of reference than that of our Gazebo visualization.
Previously this was remedied by publishing an intermediate static transform that would apply to all point cloud messages coming in, performing a series of rotations and a vertical offset to match the location and orientation of the camera. This approach will quickly fall out of favor, however, as we move towards using the stack with a real ZED camera rather than our spoofed simulation point clouds, as we would needlessly apply the transforms to the point cloud messages coming from the ZED camera. 
To address this, we need to alter the Opennni Kinect source files to rectify the frame of reference, and properly include these in the build pipeline.

arschallwig, 2022-10-02:
Custom libraries now integrated into build pipeline in `src/gazebo/gazebo_ros_...` 
Potential source of disparity in `src/gazebo/gazebo_ros_openni_kinect.cpp` or [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/231a7219b36b8a6cdd100b59f66a3df2955df787/gazebo_plugins/src/gazebo_ros_openni_kinect.cpp#L364) - where is this urdf defined? Is this different from the `rover_gazebo_plugins.xacro` that we define?

zwgold, 2023-01-19:
Closing this since we fixed it with PR #189.

closed 2023-01-19 by zwgold