umrover/mrover-ros #265: Fix differential drive

tabiosg, 2022-11-21:
This code is incorrect.
https://github.com/umrover/mrover-ros/blob/master/src/esw/drive/odrive_control.py#L654

When moving and turning at the same time, it is possible for the outer wheels for one side to move in the opposite direction of the inner wheels for that same side.

Our equation simplified looks like

```python
left_rad_inner = (forward - turn_difference_inner)
left_rad_outer = (forward - turn_difference_outer)
```

It is possible for turn_difference_inner to be less than forward but turn_difference_outer to be greater than forward. Thus, the left_rad_outer will be negative while the left_rad_inner will be positive.

Someone should fix the logic (help needed).

ankithu, 2023-01-19:
status?

tabiosg, 2023-01-19:
i didnt see anything wrong with it soooo ya

WinsonChen114, 2023-01-20:
The logic should be correct, if the rover is turning, it should be turning around a point, and all the wheels on one side of the point will be moving in the same direction. Wheels on the other side of the point will be moving in opposite directions. If we look at turning in place, the center of rotation is the center of the rover, so we can say that the displacement from the center of mass to the center of rotation is 0, and the displacement to the right wheels are negative, and the displacement to the left wheels are positive. So, the left and right wheels rotate in opposite directions.

As the center of rotation moves left, so the rover makes a bigger and bigger left turn (and the displacement to the center of rotation becomes a bigger and bigger positive number), there will be a point when the displacement to the center of rotation is greater than the left inner wheel, but not as great as the left outer wheels, because the outer wheels are further from the center of the rover than the inner wheel. This means that the left inner wheel and right wheels will rotate in one direction, and the left outer wheels will rotate in the opposite direction, then, as the center of rotation moves even more left, the displacement to it will be greater than all wheels, so the wheels will move in the same direction