umrover/mrover-ros #289: measure GPS gaussian noise

rbridges12, 2023-01-01:
In order to accurately simulate the GPS, we need to figure out its Gaussian noise characteristics. We will record GPS data while the GPS is stationary, and then we need to use this data to calculate the standard deviation of the Gaussian noise in the latitude, longitude, and altitude axes (altitude might not be necessary).

Here are docs for the [GPS gazebo plugin](http://wiki.ros.org/hector_gazebo_plugins)

niwhsa9, 2023-01-01:
aren't the static noise characteristics completely different from what we see during motion though?

rbridges12, 2023-01-01:
You're right, thats what we observed last year in some tests. I'm still a little skeptical of those results, but we should definitely test both. I'll see if I can think of some ways to measure noise during motion, or if you have any ideas I'm all ears

zwgold, 2023-01-23:
Do you have future dates for testing this, so we can close this out sometime?

rbridges12, 2023-01-23:
I have people working on it, no dates for testing yet

rbridges12, 2023-07-22:
this should be done for RTK this year so we can bound heading accuracy