umrover/mrover-ros #315: add locks to callback functions in misc python nodes

rbridges12, 2023-01-17:
We haven't been using locks to ensure thread safety in our rospy nodes, but its probably a good practice for callback functions that have non atomic operations on shared state. we should look through all of our python nodes that are subscribers and add locks on shared state where necessary

qhdwight, 2023-01-23:
I thought all callbacks were atomic b/c ros::spin or ros::spinOnce calls them in order?

rbridges12, 2023-01-23:
thats how it works in cpp, but python has a different thread for every cb function according to this forum post: [1](https://answers.ros.org/question/267960/callback-models-for-roscpp-and-rospy/)

qhdwight, 2023-01-24:
That's quite goofy

rbridges12, 2023-06-08:
it seems like for whatever reason this is not necessary so we probably shouldn't do it

closed 2023-06-08 by rbridges12