umrover/mrover-ros #385: Measure rover process noise covariance matrix

rbridges12, 2023-03-16:
To tune our kalman filter (EKF), we need to estimate the rover's process noise covariance matrix, which essentially tells us how much noise is caused by propagating our dynamics forward. This can probably be measured by driving the rover along a known trajectory, and measuring the error. This can be done in sim to get an idea of the general structure of the matrix/what the covariances look like, and it can be done IRL to get a more accurate measure of noise. We can also look at literature to see what process noise matrices are commonly used in EKFs for differential drive robots.

niwhsa9, 2023-03-16:
should this be bundled up with #289 ?

rbridges12, 2023-07-22:
probably not super important but still would be nice to do this for some rigor