umrover/mrover-ros #386: Refactor GPS linearization

rbridges12, 2023-03-16:
@niwhsa9 has some suggestions for refactoring the GPS linearization architecture to make it less confusing

rbridges12, 2023-03-16:
Looks good overall, why do you want the linearization function to take in ROS messages instead of numpy arrays? also a few things about covariance:
- we don't really need to input or output covariance from `compute_map_to_odom` since covariance is only used by the EKF and calling that function means we aren't using the EKF
- where should the configured GPS covariance matrix be added? should we just replace the existing one in the GPS message in the callback?

niwhsa9, 2023-03-16:
> we don't really need to input or output covariance from compute_map_to_odom since covariance is only used by the EKF and calling that function means we aren't using the EKF
:+1: 
> where should the configured GPS covariance matrix be added? should we just replace the existing one in the GPS message in the callback?
yes. Maybe its better to hvae publish_pose take in the Imu GPS messages as args for clarity, up to you.

ankithu, 2023-03-16:
would it make sense to synchronize the IMU and GPS messages using `message_filters.TimeSynchronizer` to simplify the code further so that we could just have `callback(imu_msg, gps_msg)`?

rbridges12, 2023-03-16:
> would it make sense to synchronize the IMU and GPS messages using `message_filters.TimeSynchronizer` to simplify the code further so that we could just have `callback(imu_msg, gps_msg)`?

I don't think so, because I'm pretty sure in reality the IMU and GPS have quite different refresh rates, so syncing them would lower the effective refresh rate of the IMU, and might require some extra logic to manage timstamps

ankithu, 2023-03-16:
> I don't think so, because I'm pretty sure in reality the IMU and GPS have quite different refresh rates, so syncing them would lower the effective refresh rate of the IMU, and might require some [...]

Ok makes sense

niwhsa9, 2023-03-16:
> message_filters.TimeSynchronizer

I did not know about that. Neat

PR #377 (2023-03-22): Global EKF
## Summary
Closes #386 

- Global EKF using robot_localization package that fuses commanded velocity, linearized GPS, IMU orientation, as well as IMU accelerometer, gyro, and mag (those 3 are turned off for now)
- Infrastructure to configure covariance matrices of each sensor measurement from the IMU and GPS
- GPS linearization refactor into linearization and passthrough filter nodes
- Misc sim infrastructure including ground truth path in RViz
- EKF testing script to give metrics and help tune EKF in sim

### Did you add documentation to the wiki?
Will do before merging

## How was [... cut]
closed 2023-03-22 by rbridges12