umrover/mrover-ros #516: Write CAN messages for non-moteus devices

tabiosg, 2023-07-15:
Figure out how to write CAN messages for non-moteus devices.

tabiosg, 2023-07-15:
raw_message = moteus.Command()
        raw_message.raw = True
        raw_message.destination = 32
        raw_message.bus = 5
        raw_message.reply_required = True

        # 80: reply + we are ID 0
        # ff: ID of the power dist
        raw_message.arbitration_id = 0x80ff

        # 01: write a single int8
        # 00: start register number "Mode"
        # 00: State mode
        raw_message.data = b'010000'

        results = await transport.cycle(
            [raw_message],
            force_can_check = (1 << 5),
        )

try something like this

qhdwight, 2023-07-22:
What device are you trying to communicate with out of curiosity?

tabiosg, 2023-07-22:
The custom brushed motor controller that we will eventually write.

qhdwight, 2023-07-22:
Cool. We should make a design interface that describes every CAN message and its contents.

tabiosg, 2023-07-22:
https://github.com/mjbots/pi3hat/blob/master/lib/python/examples/raw_can_example.py

tabiosg, 2023-10-10:
Addressed by @owenpark8 bless your soul

closed 2023-10-10 by tabiosg