umrover/mrover-ros #537: Get ARTag Detection Working on USB Camera

alisonryckman, 2023-09-19:
Use OpenCV to detect ARTags using the new Arducam USB camera
https://www.arducam.com/faq/opencv-v4l2-python-rpi/

qhdwight, 2023-09-25:
Use https://docs.opencv.org/4.2.0/d8/dfe/classcv_1_1VideoCapture.html

qhdwight, 2023-09-25:
I would also make this a separate node instead of being inside of perception

alisonryckman, 2023-10-01:
Because the usb_cam ros node would need to be completely rewritten to be made into a nodelet, instead let's try to use cv::VideoCapture as it should be less complicated https://docs.opencv.org/4.x/dd/d43/tutorial_py_video_display.html

MarshallVielmetti, 2023-10-01:
Sounds good - did that recently for a project and it worked well. I will read up on it then take a look during Tuesday's meeting

alisonryckman, 2023-10-03:
1. Read in from Gstreamer in YUY2 format
2. Publish YUY2 Format to a separate topic for Quintin's camera streamer node (this should be a grayscale img ROS type)
3. Also publish for percep purposes in BGRA
4. Read in image data and publish in same thread
5. Post publishing the boost shared ptr image, do not do any additional computation with it to prevent interference with other threads
6. Bitwise copy g cv::Mat image object into ROS sensor_msgs::Image to prepare for publishing
7. Buffer swapping now is likely uneccessary because image processing is not a very computationally expensive process