umrover/mrover-ros #547: Figure out solution to making brushless units consistent

tabiosg, 2023-09-25:
Drive likes to think in terms of rad/s (aka motors need to be controlled in revolutions/s).
Arm likes to think in a scale of 0 - 1. We need to figure out who handles the conversions.

My current thought is that the Brushless.set_desired_speed accepts rev/s and the BrushedMotors.set_desired_speed accepts 0-1 which represents raw PWM. But I'm not too sure what the best solution is.

qhdwight, 2023-09-28:
I think we always want to err on the side of sending physical units. For example rad/s.

Reasoning: If you switch the hardware you don't want the speed to change. This is true if you just use throttle 0-1. Feel free to debate me though.

tabiosg, 2023-09-28:
OK let me entertain the idea that we use rad/s. Does teleop send in terms of rad/s or does esw do it

qhdwight, 2023-09-28:
Teleop would send rad/s

tabiosg, 2023-09-28:
Ok fair. now we just need an elegant way to make both the firmware and teleop always be consistent with each other on what the actual limits of the motors and outputs are. Basically teleop wants to control the joint. ESW wants to control the motor. We need to make sure that the mapping is correct. 

Biggest question: Should ESW also try to figure out what the max speed of the actual motor can be? Or should we just have the control algorithm deal with it when it tries to figure out the max.

tabiosg, 2023-09-28:
Also, do you think ESW should just send out CAN messages in terms of rad/s of the motor or rad/s of the joint?

tabiosg, 2023-10-10:
Addressed by @qhdwight @xia-steven @edurso and @Jaccob-B , thank you. The answer is to send out CAN messages that are SI units of joint output.

closed 2023-10-10 by tabiosg