umrover/mrover-ros #554: Interface with moteus controller using C++

tabiosg, 2023-09-28:
Here is where to start:

https://jpieper.com/2023/09/19/c-client-bindings-for-moteus/
https://github.com/mjbots/moteus/blob/main/lib/python/examples/simple.py
https://github.com/mjbots/moteus/blob/main/lib/cpp/examples/simple.cc

tabiosg, 2023-10-10:
Ok I want to give more specific action plans: I want to just out what type of raw message I should send to send a raw velocity/torque/position command. I want to know what the format of the ID and data frame is.

tabiosg, 2023-10-10:
From this, the brushless.cpp files of the motor library can be implemented.

tabiosg, 2023-10-10:
I updated the description of this to be more clear.

qhdwight, 2023-11-15:
Had success with this: https://github.com/mjbots/moteus/tree/d77f76addb77632c4c83f84675253b17d9447998/lib/cpp/mjbots/moteus

Only hiccup was monitoring for when the Moteus sends back a position timeout, then needing to send a stop command back:

```c++
 void BrushlessController::processCANMessage(CAN::ConstPtr const& msg) {
        assert(msg->source == mControllerName);
        assert(msg->destination == mName);
        auto result = moteus::Query::Parse(msg->data.data(), msg->data.size());
        ROS_INFO("%3d p/v/t=(%7.3f,%7.3f,%7.3f)  v/t/f=(%5.1f,%5.1f,%3d)",
                 result.mode,
                 result.position,
                 result.velocity,
                 result.torque,
                 result.voltage,
                 result.temperature,
                 result.fault);

        if (result.mode == moteus::Mode::kPositionTimeout || result.mode == moteus::Mode::kFault) {
            moteus::CanFdFrame stopFrame = mController.MakeStop();
            mDevice.publish_moteus_frame(stopFrame);
            ROS_WARN("Position timeout hit");
        }
    }
    ```

closed 2023-11-15 by qhdwight