umrover/mrover-ros #560: Convert DriveControls/JoystickValues to Django

neuenfeldttj, 2023-10-03:
We're starting to convert components to include Django. We'll start with the DriveControls component (what we did in the starter). This year we are getting rid of the `jetson_teleop` node. So basically, the DriveControls component will process the joystick values and then publish the wheel output values to a `/cmd_vel` topic (see [this](https://github.com/umrover/mrover-ros/blob/fe08488535047b736ba1064e06371e81b3135890/src/teleop/jetson/jetson_teleop.py#L4) for the original `jetson_teleop` code). 

JoystickValues is just a component the reads the joystick values and displays them.

qhdwight, 2024-02-01:
@neuenfeldttj can we close this?

neuenfeldttj, 2024-02-07:
Yes

closed 2024-02-07 by neuenfeldttj