umrover/mrover-ros #573: Read Data from Spectral Sensors Using I2C Mux

tabiosg, 2023-10-10:
Description:
This GitHub issue is created to track and document the process of reading data from spectral sensors via an I2C multiplexer (mux). The primary goal is to set up the hardware and software infrastructure required to access and retrieve spectral data from these sensors, which are connected through the I2C multiplexer.

Scope:
This task encompasses the following key areas:

I2C Mux Setup:
Verify the hardware setup, including the connection of the I2C multiplexer to the spectral sensors.
Configure the I2C multiplexer to manage multiple I2C sensor addresses effectively.

Spectral Sensor Communication:
Develop firmware to communicate with the spectral sensors through the I2C multiplexer.
Ensure that the correct I2C addresses are selected for each sensor.
Retrieve spectral data from the sensors using the I2C communication protocol.

Steps to Reproduce:
Provide detailed, step-by-step instructions for setting up the I2C multiplexer, configuring the sensors, and reading data from the spectral sensors.
Include code examples, configuration parameters, and hardware connections necessary for setup.

Expected Results:
Clearly define the expected outcomes for each aspect of the task. What should be the result of successful spectral data retrieval using the I2C multiplexer?

Actual Results:
After conducting the tests and implementing the setup, report the actual results, documenting any issues, errors, or unexpected behavior encountered during the process.

Attachments:
If applicable, attach relevant code snippets, configuration files, and documentation used during the setup and implementation. This will assist others in diagnosing and addressing any issues.

Additional Information:
Include any supplementary context, unique requirements, or specific goals related to this issue. Share any particular concerns or challenges that should be addressed during the process of reading data from spectral sensors via the I2C mux.

By providing a detailed and clear description, you'll enable effective collaboration and troubleshooting within the GitHub community, ensuring successful spectral data retrieval using the I2C multiplexer.

wisabel0, 2023-10-21:
Here is the datasheet for the [I2C mux]((https://cdn-learn.adafruit.com/downloads/pdf/adafruit-pca9546-4-channel-i2c-multiplexer.pdf )) that we'll use.

qhdwight, 2024-03-03:
@wisabel0 can this be closed?

wisabel0, 2024-03-05:
This needs to stay open because we haven't been able to get accurate data from all 3 sensors yet