umrover/mrover-ros #602: Investigate dataset or combination of datasets that provide good training for mallet and water bottle

qhdwight, 2023-10-31:
The current yolov8 model we are using is trained to recognize labels present in https://tech.amikelive.com/node-718/what-object-categories-labels-are-in-coco-dataset/. It seems this does not have good enough labels for our purposes.

We should find a dataset that we can train the yolov8 model on which provides better labels.

qhdwight, 2023-10-31:
Perhaps take our own picutres and crowd source with https://roboflow.com/ ?

umroverPerception, 2023-10-31:
Attempting YOLO on [OpenImagesV7](https://docs.ultralytics.com/datasets/detect/open-images-v7) which has both a bottle and hammer class. Need to figure out to convert dataset into YOLO compatible format, possible script [here](https://github.com/ibaiGorordo/OpenImages-Yolo-converter/blob/master/OIDtoYOLOconverter.py)

Dataset already downloaded on percep laptop with 500 images of hammers and bottles. If you do download dataset on your personal machine, don't download the full thing since its like 500GB. Use [fiftyone](https://docs.voxel51.com/user_guide/dataset_zoo/datasets.html#dataset-zoo-open-images-v7) to download a subset

closed 2024-03-03 by qhdwight