umrover/mrover-ros #662: Create Amino.vue component

neuenfeldttj, 2024-03-05:
This component is used to activate heaters for the Ninhydrin test. There is one heater per site (site is passed in as a prop) that is toggled. When toggled it will call the `science_enable_[heater_n0]` service (type: `SetBool`). Make sure the number corresponds to the correct test site (A=0, B=1, C=2). It will read out the thermistor temperature as well as the status of the heater. There is also an auto-shutoff toggle for the heater as well as a status for that. Check the ESW ICD for more details. 
NEW this year: we want to be able to take a picture of the samples being tested, so add a button that will trigger a service. Service name/type TBD.
Last year's code for [reference](https://github.com/umrover/mrover-ros/blob/tjn/teleop-starter/src/teleop/gui/src/components/Amino.vue)

qhdwight, 2024-04-03:
Updates?

neuenfeldttj, 2024-04-05:
Merged for ISH practice mission, forgot to create a PR

closed 2024-04-05 by neuenfeldttj