umrover/mrover-ros #668: Make Science Thermistor Autoshutoff Value Configurable

tabiosg, 2024-03-28:
Description
This enhancement involves multiple components of our system, including the messaging protocol, the science module firmware, and the ROS configuration and codebase. The objective is to make the autoshutoff temperature threshold for the science module's thermistor configurable at runtime rather than being a fixed firmware value. This will allow for greater flexibility in operations and experiments, accommodating a wider range of temperatures for different environments or experiments.

Part 1: Messaging Protocol Enhancement
File(s) to Modify: messaging.hpp
Modification Detail: Introduce a new message type, ConfigThermistorAutoshutoff, dedicated to configuring the autoshutoff temperature threshold. This message will contain a single float parameter representing the desired autoshutoff temperature.
Expected Behavior: The new message allows for dynamic configuration of the autoshutoff temperature threshold via CAN message.
Part 2: Science Module Firmware Update
Requirement: Implement functionality to handle the ConfigThermistorAutoshutoff message.
Implementation Detail:
Add a local variable in the firmware to store the autoshutoff temperature threshold. This variable should be modifiable at runtime.
Modify the message handling logic to update this variable when a ConfigThermistorAutoshutoff message is received.
Ensure that the logic controlling the heater's autoshutoff behavior references this variable to determine the temperature threshold.
Verification: Confirm that the firmware successfully builds with these changes and that the autoshutoff behavior correctly reflects the configured threshold.
Part 3: ROS Configuration and Node Update
File(s) to Modify: ROS parameter file (e.g., esw.yaml), Science node codebase (science.cpp)
Modification Detail:
Update the esw.yaml file to include a new parameter representing the autoshutoff temperature threshold.
Modify science.cpp to read this parameter from the ROS parameter server at startup and transmit a ConfigThermistorAutoshutoff message over CAN with the configured value.
Expected Behavior: Upon startup, the Science node reads the configured autoshutoff threshold from the ROS parameter server and communicates this value to the science module firmware via CAN.

wilsonchen02, 2024-03-29:
Part 2 done, it builds. Haven't done the ROS part yet

tabiosg, 2024-03-29:
sicko mode

PR #672 (2024-03-29): Thermistor Auto Shutoff is now configurable
## Summary
Closes #668
closed 2024-03-29 by tabiosg