umrover/mrover-ros #682: Update ROS Bridge to not call define "Radians" or "RadiansPerSecond" when not applicable

tabiosg, 2024-04-09:
Currently, we have Velocity or Position commands that we used to assume were always in Radians.
However, now, sometimes, they may be in terms of meters. In this case, we set the value "mAvoidConversionToRevolutions" to true.

Currently, we have the functions "setDesiredPosition" and "setDesiredVelocity" but they take in the arguments Radians. It might make more sense to have these functions be just raw dimensionless/float values and then convert them inside later.

For example,
```
function_name (float argument_velocity)

argument_velocity = std::clamp (argument_velocity, minvel, maxvel);
Revolutions moteus_input_command; 

if (mAvoidConversionToRevolutions) {
    moteus_input_command = Revolutions{argument_velocity};  // This treats it as the raw value.
}
else {
   moteus_input_command = Radians{argument_velocity};
}
```

Note that this needs to catkin build so this needs to work for the Brushed Controller as well.

qhdwight, 2024-04-09:
I think it makes sense for these functions to accept any templated unit type actually

qhdwight, 2024-04-09:
This may be kind of advanced metaprogramming I'll take a stab at it